PLC-lang/rusty #1194: initialization error not visible

rarris, 2024-04-09:
if we have a error by initialization, this is not shown at all - line 4
the initialization error should be first listed

```
PROGRAM mainProg
VAR
x1,x2 : INT;
ton1 : TOFF;
END_VAR
x1 := 'cc';
END_PROGRAM
_________
error[E037]: Invalid assignment: cannot assign 'STRING' to 'INT'
Compilation aborted due to critical errors
```

**2nd problem hier**
if we comment or delete "x1 := 'cc';", so that the initialization error be visible, the error is a codegen error and the link in the console is not listed

volsa, 2024-04-11:
To add some context, the "Unknown type: TOFF" error pops up in codegen, but the invalid assignment prevents us from entering codegen in the first place. We should probably detect errors like these in the validation stage rather than in codegen.

volsa, 2025-08-07:
Can no longer be produced, can you confirm @rarris