PLC-lang/rusty #1211: Provides a function that allows incremental compilation checks

XLJ707781299, 2024-04-23:
# I'm going to do a dynamic code error checking feature now
Users can enter an ST code during this process to display an error message. But during the input process, if you had to send all the ST files to rusty every time, it would obviously be slow.

# Ideas
When the user opens the ST code project for the first time, the ST code of all files is sent to rusty for inspection. In the subsequent user modification process, only the current modified file is sent to rusty. Considering that there may be references between ST code files, this will be a complex process.

# Request
Is there a good way to implement this feature? Or there is already such an interface, I just haven't found it yet

ghaith, 2024-04-23:
This would be an incremental compilation which we did not yet implement but is something we would like to eventually have.

Regarding the slowness our bottlenecks are usually at codegen so sending all files to rusty and calling `plc check` should be usually done in matters of milliseconds or seconds for extremely big projects.

ghaith, 2024-04-23:
One way to implement incremental compilation is to use the rusty command line without the build pipeline.
What you can do is have a Makefile or similar compile each file on its own, where all other files required by this file (can also be all files) are included with the `-i` flag. This will still take generally  the same parse time but the resolve and validate steps would be faster since they don't have to look into the included files' code.
This is usually not an easy process so it probably needs some automation on the Makefile side and i'm not sure how much performance you would win.

riederm, 2024-04-23:
hi @XLJ707781299m, thx for the issue first. Can you share a bit information about the size of your project (lines of code, number of files, compiler arguments ... stuff like that) and how you experience the performance of the compiler?

so far we have found that rusty performs pretty good, I think eventually we will tacke features like an incremental build but I would be very interested in the compile-times you experience in your setup.

XLJ707781299, 2024-04-24:
@riederm @ghaith  I am very glad to receive your reply. I am currently testing 10,000 files, the content is a simple ST code: 
[image]
[image]
, the total amount of data of all files is about 1.75M, use the command "plc.exe --check st\*", the time spent is about 604ms, I use the code version:
[image]
For now, this performance appears to be acceptable for automatic error checking, but this is obviously directly affected by the total data size of all files. If you have incremental compilation capabilities, you can minimize this impact.

volsa, 2024-04-24:
Regarding these files, are there any sensible information on them? If not would you mind sharing the project? We're currently in the process of improving the performance of some parts of the compiler and further projects to test it on wouldn't hurt :)