PLC-lang/rusty #1226: RuSTy fails to compile a simple function block

rohanrayan, 2024-05-22:
**Describe the bug**
I am trying to compile the following code with RuSTy and it gives an error, could you please let me know how to fix it?

```
FUNCTION_BLOCK testAddon
    METHOD PUBLIC testmethod : STRING
        VAR_INPUT
            testvar: DWORD;
        END_VAR    
    END_METHOD
END_FUNCTION_BLOCK
```

error:
```
thread 'main' panicked at src/codegen/generators/pou_generator.rs:561:22:
internal llvm codegen error: ()
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**To Reproduce**
Copy the code snippet to a .st file (ex: sample.st)
try to compile using:
`plc --ir sample.st`

**Expected behavior**
I expect RuSTy to generate an LLVM-IR for this code

riederm, 2024-06-06:
hi @rohanrayan , thx for the issue.
so far we implemented a language grammar of iec61131-3 from a version that did not support oop (classes and methods). Implementing OOP support (ant therefore also methods) is scheduled for the 2nd half of 2024.

We'll need to postpone this issue a bit, but we'll get there soon.

rohanrayan, 2024-07-29:
Hi @riederm 

Could you let me know which version (or year) of the grammar that is currently supported by RuSTy?

rohanrayan, 2025-01-07:
Is there any update on this issue? Is there a timeline when I can expect this to be fixed?

Thanks!

ghaith, 2025-01-08:
Hello @rohanrayan,
we have support for methods in FB now, but AFAIK without modifiers, could be that we just ignore the modifires.
The support is however still in its infancy we don't for example support polymorphism yet and we still have some edge cases in method calls that we want to iron out.
The version we aim to support is the 2014 version which includes OOP, we are however focusing more on the codesys variants of OOP for the start

rohanrayan, 2025-01-08:
Thanks. Is there any timeline for the completion of the OOP support?

ghaith, 2025-01-08:
We started implementation, so within the next few months. We already have methods and interfaces as mentioned, we want to support polymorphism as a next step. namespaces as defined in the norm are not yet a priority but will hopefully come later.

volsa, 2026-02-27:
Should be fixed, please re-open if still reproducible @rohanrayan

closed 2026-02-27 by volsa