PLC-lang/rusty #1241: Windows github runner broken

ghaith, 2024-06-10:
**Describe the bug**
Our tests currently fail on windows because of a change in the github runner. They run fine locally on a windows install.
I found an issue on the runner repo talking about a change in the C++ runtime, could be related but i can't be sure
https://github.com/actions/runner-images/issues/10020

We should find a more reliable way to run the windows tests that would not break with changes to the runtime.

riederm, 2024-06-10:
Can we discuss if we deacitvate the windows build-action for the moment, and re-activate it when this issue was done? 
Every PR fails atm.

ghaith, 2024-06-10:
The windows action does not block the build it just reports a failure but you can still submit. I'm also fine with ignoring the test until we fix it however.
@volsa is checking it out at the moment.

riederm, 2024-06-10:
I find it disturbing that everything shows up as :x: in the overview.
The repository's overall build status is also "failed"

ghaith, 2024-06-10:
I created a PR #1242 that splits the workflows and disabled windows for now. Feel free to submit it and we can enable it again when the issues are fixed

volsa, 2024-06-10:
FYI I wasn't able to find a solution, but as far as I can tell GitHub / Microsoft [unofficially acknowledged](https://github.com/actions/runner-images/issues/10004#issuecomment-2157555139) the issue and are working on a hotfix? Personally I would just wait this out and see if the issue fixes itself in a few days.

ghaith, 2024-06-13:
The fix is deployed, builds now run again

closed 2024-06-13 by ghaith