PLC-lang/rusty #1278: Failed to compile examples on Mac ARM64

zsx, 2024-08-10:
**Describe the bug**

It fails to compile the files under example/ on an M2 Macbook.

**To Reproduce**
Steps to reproduce the behavior:

I tried to follow the example as [documented](https://plc-lang.github.io/rusty/using_rusty.html#compiling-with-rusty):
`cargo run plc examples/hello_world.st -o hello_world --linker=cc` complains:

```
LLVM ERROR: Global variable 'main' has an invalid section specifier ',fn-$RUSTY$main:i32': mach-o section specifier requires a section whose length is between 1 and 16 characters.
```

**Expected behavior**
the command should produce an executable

**Additional context**
Add any other context about the problem here.

**Tests**
Optional: Add a link to a PR or a branch where you have a failing test of the issue.

ghaith, 2024-08-12:
Seems like the changes to the metadata affects macs. There's a PR that will move them behind a flag but we should still look into this.

volsa, 2024-08-12:
Related https://github.com/PLC-lang/rusty/issues/1192; in the meantime you'll have to use Docker / Devcontainer when developing under MacOS

volsa, 2026-09-29:
The failing section specifiers are gone from the default path: since #1319 (16b756fa5f5), `$RUSTY$` sections are only emitted with `--online-change`. `examples/hello_world.st` now compiles to a Mach-O arm64 object for `aarch64-apple-darwin`. I could only cross-compile on Linux, and `--online-change` still fails on Mach-O, so please reopen (or open a new issue for online change) if you still see a problem on a Mac.

closed 2026-09-29 by volsa