PLC-lang/rusty #1308: Support array indexing in VAR_CONFIG blocks

volsa, 2024-09-12:
**Describe the bug**
Indexing any array variable within a `VAR_CONFIG` block currently reports a parsing error, e.g.

```
VAR_CONFIG
    main.foo[1].bar AT %IX1.0 : BOOL;
END_VAR
```
will report
```
error[E006]: Missing expected Token AT
  ┌─ target/demo.st:2:12
  │
2 │     foo.bar[1] AT %IX1.0 : BOOL;
  │            ^ Missing expected Token AT

error[E006]: Missing expected Token hardware access
  ┌─ target/demo.st:2:12
  │
2 │     foo.bar[1] AT %IX1.0 : BOOL;
  │            ^ Missing expected Token hardware access

error[E007]: Unexpected token: expected KeywordSemicolon but found '[1] AT %IX1.0 : BOOL'
  ┌─ target/demo.st:2:12
  │
2 │     foo.bar[1] AT %IX1.0 : BOOL;
  │            ^^^^^^^^^^^^^^^^^^^^ Unexpected token: expected KeywordSemicolon but found '[1] AT %IX1.0 : BOOL'
```

mhasel, 2024-09-23:
In a similar vein, declaring a custom string type, e.g.
```
VAR_CONFIG
    main.foo.bar AT %IX1.0 : STRING[100];
END_VAR
```
currently leads to a panic in `index::visitor::visit_global_var_block`, `src/index/visitor.rs:313:50` due to the custom string type not having a name. Not yet sure if this is also a parser issue, if not I'll open a separate issue once I've investigated further.

mhasel, 2024-10-08:
Since parsing arrays is no longer an issue, I'm closing this issue. Custom string type panic tracked in #1331

closed 2024-10-08 by mhasel