PLC-lang/rusty #1313: Automated docker images.

corbanvilla, 2024-09-19:
**Is your feature request related to a problem? Please describe.**
It would be convenient to have an auto-building docker image with a compiled version of `plc` ready. 

I have one configured with a github workflow that I would be happy to submit through a PR if it's something that would be accepted :)

ghaith, 2024-09-19:
Sure, if you submit one we can integrate it and have it push to the github container repository  or to docker hub automatically

ghaith, 2024-09-19:
For the record, we have images ready to build the compiler here: https://github.com/PLC-lang/rust-llvm-images
You can use them as base to build the image and then ship a minimal image with just plc and a linker. But it might make sense to keep a compiler in that image if you want to also link other languages.. Alternatively you could only have it start plc and have the compiler be on your system or in another image

corbanvilla, 2024-09-19:
Sounds good, I pulled in the template from there :) 

The image is a bit bloated, but it may be useful to have all the dependencies in one image, and a minimal version in another.

mhasel, 2024-10-14:
@corbanvilla @ghaith I'm closing this issue since it seems like this feature has been added with #1314. Please reopen if there is still work to do.

closed 2024-10-14 by mhasel