PLC-lang/rusty #1414: Win11 rusty build compiles hello_world sample but does not run as expected

zeehans, 2025-02-20:
Not sure if this is indeed a bug or just something got mixed up with my setup: 

I just pulled the latest master commit and rebuilt rusty on a Win11 machine. I build the hello_world example from the docs using

`plc -c main.st -o main.exe --linker=clang`

and it completes without an error. When I try to run it from powershell I get:

`Program 'main.exe' failed to run: The specified executable is not a valid application for this OS platform.At line:1`

I havent been using rusty in a while and used to use a build from an older master commit - probably almost two years old - and was able to compile and run far more complicated code.

Due to a Siemens AX trial installation I know that there was another clang toolchain installed on my system but that one is not on the path and pretty sure I am using the custom [LLVM 14.0.6](https://github.com/PLC-lang/llvm-package-windows/releases/tag/v14.0.6) toolchain.

Any ideas what could be causing this?

ghaith, 2025-02-20:
Hello @zeehans 

There are 2 things I notice:
- You use the `-c` flag but try to link the application, `-c` is compile only, which will skip linking. If you remove the flag you should not be compiling an executable for windows
- Second option that might be worth a try if removing the `-c` does not help is to divide the compile steps in 2, this is because sometimes clang can be configured to use the wrong linker to compile. I don't think it is happening in your case but it's worth a try:
```
plc -c main.st -o main.o
clang main.o -fuse-ld=lld-link -o main.exe
```
This will force clang to use the lld-link linker to link the executable.

ghaith, 2025-02-21:
could it be that oldnames was removed for win11? 
Let's try another approach, can you compile with 
`clang main.o -o main.exe` skipping the lld-link part? this will link with the msvc linker (`link.exe`) let's see if it does a better job. You can also force it with `-fuse-ld=link`
I don't have a windows 11 machine at the moment but i'll try doing more tests later with that.

zeehans, 2025-02-21:
@ghaith just tried your suggestions with `-fuse-ld=link` and without and in either case the error message is

```
clang: error: unable to execute command: program not executable
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```