PLC-lang/rusty #1473: Help getting debugging information in program POUs?

thetooth, 2025-05-02:
I'm building my project using the project.json approach as an object file and linking it into a runtime I'm writing in C++, everything works fine however when setting a break point inside the ST code none of the locals (anything defined inside VAR) are visible to the debugger, everything that's an input or has memory defined in the C++ project is visible.

[image]

ghaith, 2025-05-02:
hello @thetooth , because of how a `PROGRAM` is saved in memory, a local variable contained in a `VAR` block is actually global, `timeOut` is accessible as `PLC_PRG.timeOut`.
Local/Stack allocated variables are also supported, they would then be `VAR_TEMP` in `PROGRAM`s and `FUNCTION_BLOCK`s and `VAR` in `FUNCTION`s. 
I'm not sure if we can mark the variables in a different way for the debug adapter (vs code in your case) would see them correctly. But the variables should certainly be available in the variables view under `PLC_PRG`

thetooth, 2025-05-02:
I can't seem to access it globally either:
```
image lookup -v -s PLC_PRG
1 symbols match 'PLC_PRG' in /home/user/ethercat-toy/build/ethercat-toy:
        Address: ethercat-toy[0x00000000000aead0] (ethercat-toy.PT_LOAD[1]..text + 533312)
        Summary: ethercat-toy`PLC_PRG at PLC_PRG.st:24
         Module: file = "/home/user/ethercat-toy/build/ethercat-toy", arch = "x86_64"
    CompileUnit: id = {0x00000008}, file = "/home/user/ethercat-toy/src/iec/PLC_PRG.st", language = "c"
       Function: id = {0x0033daf9}, name = "PLC_PRG", range = [0x0000555555602ad0-0x0000555555602cd0)
       FuncType: id = {0x0033daf9}, byte-size = 0, decl = PLC_PRG.st:8, compiler_type = "void (void)"
         Blocks: id = {0x0033daf9}, range = [0x555555602ad0-0x555555602cd0)
      LineEntry: [0x0000555555602ad0-0x0000555555602b45): /home/user/ethercat-toy/src/iec/PLC_PRG.st:24
         Symbol: id = {0x000030ba}, range = [0x0000555555602ad0-0x0000555555602cd0), name="PLC_PRG"

image lookup -v -s PLC_PRG.timedOut
(it's blank)
```
```
target variable PLC_PRG
(PLC_PRG) PLC_PRG = {
  tick = 799493372358694400
  t1 = (IN = false, PT = 1, Q = false, ET = 1, __signal__ = false, __is_running__ = true, __BUFFER__ = "в\0\0\0\0\U00000001\U00000001\0\0\0\0\0\0y\U00000010\U00000001\0\0\0\0\0\x9aF")
   = 0
  t2 = (IN = false, PT = 0, Q = false, ET = 1, __signal__ = false, __is_running__ = false, __BUFFER__ = "\0\0\0\0\0\0\0\U00000001\0\0\0\0\0\0}\U00000010\U00000001\0\0\0\0\0\U00000015O")
  enable = false
  pumpRun = false
  timedOut = false
  a = -12.942627
  minimum = 3.9236357E-44
  maximum = 0
}
```

In the above case the timedOut variable is actually set to true, all of the data shown is random each time the break point is hit suggesting it's looking at memory in the wrong place.

tisis2, 2025-05-05:
also see https://github.com/PLC-lang/rusty/pull/1468
it seems the wrong value you see could be caused by the same this changed was to fix

tisis2, 2025-05-05:
are you sure you running the current source when debugging? it seems very strange that e.g. a tick and t1 are shown as members of PLC_PRG but they are not declared in your source that is shown

mhasel, 2025-05-12:
The random data display should be fixed as soon as https://github.com/PLC-lang/rusty/pull/1472 is merged

thetooth, 2025-05-12:
@mhasel neat! I'll test it later this week.