PLC-lang/rusty #1475: `THIS` should not be in codegen

abroooo, 2025-05-09:
With the current implementation we create a pointer for `THIS` in codegen. While this works fine it just might be a bit overkill because we could also just resolve all references to `THIS` in the resolver.

The current implementation doesn't hurt though because the created pointer gets removed by llvm anyways. It just would be cleaner.

mhasel, 2025-05-12:
There is no local variable for the `%0` argument, we only allocate GEPs for members behind the pointer. Since this needs to be a pointer but `%0` is a `StructValue` in codegen, we need this alloca. LLVM does not allow to take the address of a register value directly, so we need to allocate space in a register to store the address.
If you can figure out a solution without the alloca, I'm all for it, but I haven't had success when I tried.

ghaith, 2025-05-12:
The idea is that this is only a marker, there is no need for it to be part of codegen as far as I can tell, it should only help us find resolve the correct variable to resolve. Do you see it differently? The only thing I can think about is calling `THIS^()` within a method, but this is  very specific. Adding the pattern to Codegen IMHO is not necessary at this point.

mhasel, 2025-05-12:
So the idea is to scrap supporting `THIS^()`?