PLC-lang/rusty #1481: Does Rusty support running on the WIN7 system?

XLJ707781299, 2025-06-04:
Hello, I compiled rusty into a dll and linked and called it through the main exe program. The compilation command for rusty is "cargo build --release --target=i686-pc-windows-msvc". When the program starts on the WIN7 system, it will encounter the problem shown in the picture. How can this problem be solved?

[image]

ghaith, 2025-06-04:
Hello, we don't actively do anything to stop win7, but we also don't actively do anything to support it either.
Since win7 has not been maintained for a while now, it is possible that our dependencies are not compatible.

If you can have better debug information from the error, like for example a trace, you might be able to find exactly which dependency misbehaves, but I'm afraid we won't be able to help you there. If you do find the issue, and the fix does not break win11 compatibility we could review a pull request.

ghaith, 2025-07-03:
Closing this issue as not planned.
If you do eventually get it running on win7 and have a PR to update the documentation i'll happily review it.

closed 2025-07-03 by ghaith